ETrobocon/etrobo #10: sample_c4が存在しない？

NaohikoYamaguchi, 2020-06-16:
担当者様

山口と申します。

ETロボコンWebサイトに記載の手順
https://github.com/ETrobocon/etrobo/wiki/sim_win

を参考に、VSCodeを含めてWindows10に環境をセットアップしました。
インストールは無事に完了し、VSCodeからUbuntuにアクセスすることもできていますが、肝心の
make sample
を実行しても、テスト環境が構築されません。

ETロボコン環境を立ち上げ直し、updateコマンドを実行して環境のアップデートをしたうえで行ってもだめでした。

エラーメッセージは以下の通りです。
`nycity@MyComputer:~/etrobo$ make sample
cp: 'sample_c4/Makefile.inc' を stat できません: そのようなファイルやディレクトリはありません
/home/nycity/etrobo/scripts/fakemake.sh: 行 90: sample_c4/Makefile.inc: そのようなファイルやディレクトリはありません
cp: 'sample_c4/Makefile.inc' を stat できません: そのようなファイルやディレクトリはありません
/home/nycity/etrobo/scripts/fakemake.sh: 行 94: sample_c4/Makefile.inc: そのようなファイルやディレクトリはありません
invoker make app=sample_c4 sim up
../common/Makefile.workspace:94: /home/nycity/etrobo/hrp3/sdk/workspace/sample_c4/Makefile.inc: そのようなファイルやディレクトリはありません
make: *** ターゲット '/home/nycity/etrobo/hrp3/sdk/workspace/sample_c4/Makefile.inc' を make するルールがありません.  中止.
fakemake on HRP3: make failed ... app=sample_c4
mv: 'sample_c4/Makefile.inc.org' を stat できません: そのようなファイルやディレクトリはありません`

そこで、etrobo/hrp3/sdk/workspace を探ってみたところ、
そもそもworkspaceの中にsample_c4ディレクトリが存在していませんでした。
`ycity@MyComputer:~/etrobo/hrp3/sdk/workspace$ ls -a
.   Makefile    gyroboy   periodic-task  test-cyc         trike
..  currentapp  helloev3  test-cpp       test-spp-master`

私だけでなく、当校の学生にも同じ症状が生じていますので、インストールパッケージに誤りがあるか、手順に何らかの抜けか誤りがあると思われます。

ご確認をお願い致します。

jtFuruhata, 2020-06-16:
ご報告ありがとうございます、これから原因を考えてみますが、ひとまず

```
cd dist
tar xvf sample_c4.tar.gz
mv sample_c4 ../workspace/
cd ..
```
でご対応よろしくお願いします。

jtFuruhata, 2020-06-16:
原因の候補に思い当たりました、もしも上記コマンドでエラーが出るようでしたら、原因はシンボリックリンクの作成失敗な気がします（インストールディレクトリがNTFSではないなど）、エラーメッセージと`etrobo`ディレクトリの`ls -la`の結果をお知らせください、よろしくお願いします。

NaohikoYamaguchi, 2020-06-16:
ありがとうございます。

さっそく試してみたところ、

圧縮ファイルの解凍まではうまくいきましたが、workspaceへの移動で「許可がありません」のエラーが出ます。
sudoをつけて実行してみましたが結果は同じでした。

```
nycity@MyComputer:~/etrobo$ cd dist/
nycity@MyComputer:~/etrobo/dist$ ls
etrobosim2020_0.992_linux.tar.gz  etrobosim2020_0.992_win.tar.gz  sample_c4.tar.gz
etrobosim2020_0.992_mac.tar.gz    sample_c4
nycity@MyComputer:~/etrobo/dist$ tar xvf sample_c4.tar.gz
sample_c4/
sample_c4/Makefile.inc
sample_c4/app.c
sample_c4/app.cfg
sample_c4/app.h
nycity@MyComputer:~/etrobo/dist$ mv sample_c4 ../workspace
mv: 'sample_c4' から '../workspace/sample_c4' へ移動できません: 許可がありません
nycity@MyComputer:~/etrobo/dist$ sudo mv sample_c4 ../workspace
[sudo] nycity のパスワード: 
mv: 'sample_c4' から '../workspace/sample_c4' へ移動できません: 許可がありません
nycity@MyComputer:~/etrobo/dist$ 
```

etroboディレクトリの状況は以下の通りです。
```
nycity@MyComputer:~/etrobo$ ls -la
合計 20
drwxrwxrwx 1 nycity nycity  4096  6月 16 14:42 .
drwxr-xr-x 1 nycity nycity  4096  6月 16 14:50 ..
drwxrwxrwx 1 nycity nycity  4096  6月 16 14:50 .git
-rw-rw-rw- 1 nycity nycity   476  6月 16 14:39 .gitignore
-rw-rw-rw- 1 nycity nycity  1086  6月 16 14:39 LICENSE
-rw-rw-rw- 1 nycity nycity 15970  6月 16 14:39 README.md
drwxrwxrwx 1 nycity nycity  4096  6月 16 14:42 athrill
drwxrwxrwx 1 nycity nycity  4096  6月 16 14:42 athrill-gcc-package
drwxrwxrwx 1 nycity nycity  4096  6月 16 14:42 athrill-target-v850e2m
drwxrwxrwx 1 nycity nycity  4096  6月 16 14:50 dist
drwxr-xr-x 1 nycity nycity  4096  4月 14 16:19 ev3rt-1.0-release
drwxrwxrwx 1 nycity nycity  4096  6月 16 14:43 ev3rt-athrill-v850e2m
drwxrwxrwx 1 nycity nycity  4096  6月 16 14:41 gcc-arm-none-eabi-6-2017-q1-update
drwxr-xr-x 1 nycity nycity  4096  4月 14 16:19 hrp3
-rwxrwxrwx 1 nycity nycity    52  6月 16 14:39 make
drwxrwxrwx 1 nycity nycity  4096  6月 16 14:39 scripts
-rwxrwxrwx 1 nycity nycity    61  6月 16 14:39 update
lrwxrwxrwx 1 nycity nycity    18  6月 16 14:41 workspace -> hrp3/sdk/workspace
nycity@MyComputer:~/etrobo$ 
```

ちなみに、シンボリックリンクの飛び先であるhrp3/sdkのls- laは以下の通りです。
```
nycity@MyComputer:~/etrobo/hrp3/sdk$ ls -la
合計 24
drwxr-xr-x 1 nycity nycity  4096  6月 16 14:41 .
drwxr-xr-x 1 nycity nycity  4096  4月 14 16:19 ..
-rw-r--r-- 1 nycity nycity 11358  4月 14 16:19 LICENSE
-rw-r--r-- 1 nycity nycity   165  4月 14 16:19 Makefile.sdk.conf
-rw-r--r-- 1 nycity nycity    16  4月 14 16:19 README.md
drwxr-xr-x 1 nycity nycity  4096  4月 14 16:19 common
drwxr-xr-x 1 nycity nycity  4096  6月 16 14:41 firmware
drwxr-xr-x 1 nycity nycity  4096  6月 16 14:51 workspace

```

インストール先のハードディスクはNTSCでフォーマットされています。

jtFuruhata, 2020-06-16:
ちなみに、777にすると問題なく書き込めますか？
（755になっているのはTOPPERS/EV3RTのオリジナルがそうなっているからで、
　sample_c4自体は777で固めてあります、というぐらい、やぶさかではないと思っています）

NaohikoYamaguchi, 2020-06-16:
試してみましたが、ダメでした。

chmod自体には成功したのですが
```
nycity@MyComputer:~/etrobo/hrp3/sdk$ sudo chmod 777 workspace/
nycity@MyComputer:~/etrobo/hrp3/sdk$ ls -la
合計 24
drwxr-xr-x 1 nycity nycity  4096  6月 16 14:41 .
drwxr-xr-x 1 nycity nycity  4096  4月 14 16:19 ..
-rw-r--r-- 1 nycity nycity 11358  4月 14 16:19 LICENSE
-rw-r--r-- 1 nycity nycity   165  4月 14 16:19 Makefile.sdk.conf
-rw-r--r-- 1 nycity nycity    16  4月 14 16:19 README.md
drwxr-xr-x 1 nycity nycity  4096  4月 14 16:19 common
drwxr-xr-x 1 nycity nycity  4096  6月 16 14:41 firmware
drwxrwxrwx 1 nycity nycity  4096  6月 16 14:51 workspace
```
移動しようとすると、やはり止められてしまいます。
```
nycity@MyComputer:~/etrobo/dist$ mv sample_c4 ../workspace
mv: 'sample_c4' から '../workspace/sample_c4' へ移動できません: 許可がありません
nycity@MyComputer:~/etrobo/dist$ sudo mv sample_c4 ../workspace
mv: 'sample_c4' から '../workspace/sample_c4' へ移動できません: 許可がありません
nycity@MyComputer:~/etrobo/dist$ sudo mv sample_c4 ../hrp3/sdk/workspace/
mv: 'sample_c4' から '../hrp3/sdk/workspace/sample_c4' へ移動できません: 許可がありません
```

ふと思いついて、VSCodeを一度終了し、
デスクトップのStartETrobo.cmdを管理者モードで立ち上げてみましたが
変化なしでした。

jtFuruhata, 2020-06-17:
ごめんなさい説明が悪かったです、`workspace`シンボリックリンク自体は最初から777ですので、
`chmod 777 "$ETROBO_ROOT/hrp3/sdk/workspace"`を試してみていただきたいです、ご協力よろしくお願いします。

NaohikoYamaguchi, 2020-06-17:
私こそわかりにくくてごめんなさい。
シンボリックリンクの先を777にしましたがダメでした。
`nycity@MyComputer:~/etrobo/hrp3/sdk$ sudo chmod 777 workspace/`

shin3505, 2020-06-17:
Heliosの小嶋と申します。
私も同じく状況(sample_c4がworkspaceにない）に陥っています。
PCはWindows10 Pro 64bit版です。

chmodの件も同じく「Permmison Denied」になっています。
lsでアクセス権確認しても「workspace」フォルダは既に777の設定になっています。
なので、まったく同じ状況で止まっています。

ちなみにですが、エクスプローラーで開いてsample_c4フォルダをworkspaceフォルダに移すことはできました。この時に、lsコマンド使ってアクセス権確認しましたが777になっていたので行けるかなと思い「make sample」しましたが別の要因でダメでした。

それが以下の画面になります。gccのコンパイラが入ってないと思われます。
ここら辺のインストールはシミュレーターインストール時に自動でやってくれるものと認識していますが違いますか？
[image]

jtFuruhata, 2020-06-17:
あああ失礼しました、その操作になっていましたね。
これはもう、ちょっと、Zoomで山口さんの環境を画面共有していただいて、こちらでも見ながら考え付く限りの対応を試してみるしかないかなと思っているのですが、よろしいでしょうか？
他にも、WSL側から/mnt/c/側のWindowsアプリを起動できない個体があるなど、意外とWSL1環境には個体差があるようなんです。
明日は10:00～17:00まで開けていますので、ご都合付くようでしたらよろしくお願いします。
 https://zoom.us/j/96964735660?pwd=Vko0cEtsb1pYbkl6WXFheVl6RHIxQT09

NaohikoYamaguchi, 2020-06-18:
丁寧にご対応ありがとうございます。

> 明日は10:00～17:00まで開けていますので、ご都合付くようでしたらよろしくお願いします。
> https://zoom.us/j/96964735660?pwd=Vko0cEtsb1pYbkl6WXFheVl6RHIxQT09

15時ぐらいでしたら、時間が作れそうです。よろしいでしょうか。
上記Zoomに直接伺えばよろしいですか？

jtFuruhata, 2020-06-18:
はい、直接で大丈夫です、開きっぱなしにしてありますので、ご協力よろしくお願いします。

NaohikoYamaguchi, 2020-06-18:
ありがとうございます。

jtFuruhata, 2020-06-18:
ご協力ありがとうございました、対応してみましたので、恐れ入りますがetroboパッケージの再インストールして試してみてください：

- Ubuntuターミナルを開く
- `./startetrobo clean`
- `exit`
- デスクトップの`Start ETrobo.cmd`をダブルクリック

これでうまくいくんじゃないかと思います、よろしくお願いします。

jtFuruhata, 2020-06-18:
お待たせしてすみません、小嶋さんの件これから追ってみます。

jtFuruhata, 2020-06-18:
小嶋さん、gcc-armはmvで展開していませんので別問題のようでした。すみませんがご協力お願いします。

まず`~/etrobo`ディレクトリ上から以下のコマンドの実行結果を教えてください：
```
echo $ETROBO_ROOT
echo $ETROBO_HRP3_GCC_URL
ls | grep gcc-arm
```

または、上記のアドレスで17:00までZoom開いています。その後、同じアドレスで20:00以降も（眠たくなるまで）開いておきますので、よろしければこちらの方が手っ取り早いです（10分もかからず解決できると思います）。

NaohikoYamaguchi, 2020-06-18:
山口です。
テスト環境を削除したのち、再度セットアップを行って、
無事ストレートにテスト環境が立ち上がることを確認いたしました。

ありがとうございました。

jtFuruhata, 2020-06-18:
ちょっと先回りして心当たりをつぶしてみました。
もしも`ls | grep gcc-arm`で何も表示されないようであれば、何らかの理由でダウンロードに失敗している（ほんとgcc-armのダウンロードは先方のネットワーク都合で重たいんですよねぇ…）ので、この場合にはエラーメッセージを表示してインストーラーを停止させるようにしました。

この場合、再インストールが必要ですので、Windowsの場合は

- Ubuntuターミナルを開く
- `./startetrobo clean`
- `exit`
- デスクトップの`Start ETrobo.cmd`をダブルクリック

をよろしくお願いします。

shin3505, 2020-06-18:
Helios 小嶋です

問い合わせした手前申し訳ありません。
環境構築に使用していたPCが諸事情により1weekほど手元から離れてしまいました。
そのため、せっかくいただいたアドバイスがすぐに実行できませんので、PC戻り次第の対応とさせてください。
ちなみにですが、土日のどちらかでもzoom対応いただくことは可能でしょうか？
難しいようでしたら平日に何とか時間を作ります。

jtFuruhata, 2020-06-18:
小嶋さん了解です、来週の土曜日ですとOSC2020 Online/Hokkaidoのイベント出展していますのでちょうどよいと思います。
手元に戻ってきましたら、上の手順をお試しください、再インストールして修正を適用すると無事インストールできるのではないかと思っています。

KITADA-Yoshitaka, 2020-06-19:
同じく、mvできない状況です。 ./startetrobo clean で再インストールしてから、再度 実行を
してみましたが、やはり、mvできないです。
　lsしてみると、
lrwxrwxrwx 1 kitada kitada        18  6月 19 12:45 workspace -> hrp3/sdk/workspace と出ますので、
フルパスでは /home/kitada/etrobo/hrp3/sdk/workspace になる。
その大元は、drwxr-xr-x 1 kitada kitada   512  6月 19 12:45 workspace なので、パーミッションからは
書けそうですが、「移動できません」「許可がありません」となってしまいます。
-------------------------------------------------------------------------------------------
kitada@ThinkPad-X1:~/etrobo/dist$ mv sample_c4 ../workspace/
mv: 'sample_c4' から '../workspace/sample_c4' へ移動できません: 許可がありません
kitada@ThinkPad-X1:~/etrobo/dist$

jtFuruhata, 2020-06-19:
mvはできません、それはパッケージの問題ではなくWSL1の問題ですので、どうしようもできません。そこでmvを使わない対応をしたので、先日の対応によってmvできるようになったわけではありません。

shin3505, 2020-06-27:
Heliosの小嶋です。
'ls | grep gcc-arm'では何も表示されませんでした。その後、指示いただいた再インストール方法を試したところmake sampleできるようになりました。
ありがとうございました。

jtFuruhata, 2020-06-27:
やはり原因はダウンロードエラーでしたね、ご確認ありがとうございました。
この辺りの利便性に関しては #13 に引き継ぎ、こちらはクローズとします。

closed 2020-06-27 by jtFuruhata