ETrobocon/etrobo #11: Macでのインストール時のエラー(Visusal Studio Code の設定が必要

khachiman, 2020-06-18:
インストール時に以下のエラーが起きました。

[image]

「code」はVlisual Studio Code

https://github.com/ETrobocon/etrobo/wiki/sim_mac

jtFuruhata, 2020-06-18:
ご報告ありがとうございます、ご指摘の件まったくその通りで、「etroboパッケージインストール前の動作環境構築手順（macOS）」の通りに行わなければいけません。
https://github.com/ETrobocon/etrobo/wiki/sim_mac
判り辛くてすみません、ドキュメントの方は鋭意整備中でして、もう少し判りやすく記述するようにいたします、もう少々お時間をください。

closed 2020-06-18 by jtFuruhata

khachiman, 2020-06-18:
すみません。ちゃんと書いてあったのですね。
そこを読んだはずだったのですが、後半を読んでいなかったようです。; ;

jtFuruhata, 2020-06-18:
余計なことは書いてあるのに肝心のところが読まれないのはダメドキュメントですよね^^;　Macに関してはスクショ入りのステップバイステップを用意中です。

jtFuruhata, 2020-06-18:
あっ、それよりもログの上の方でranlibコケ気味なのが気になります、`make sample`で正しく実行されてますでしょうか？

khachiman, 2020-06-18:
`make sample`でコンパイルは無事終了し、シミュレータ上で元気に動いています。

jtFuruhata, 2020-06-18:
お騒がせしました^^; サンプルコースでも一通り試せると思いますので、楽しんでください