ETrobocon/etrobo #17: make appが出来ません

CIT-stu, 2020-06-30:
題名の通りmake appが出来ません。

＜作業環境＞
macOS（最新版）

＜パス＞
home/BeerHall/etrobo/hrp3/sdk/workspace/sample_c4（←こいつを動かしたい）

＜エラー表示＞
コンパイル時に以下のエラーが出ます。
No rule to make target `/Makefile.fakemake'.　

そこで、workspaceに入っているMakefileを開くと以下のようになっています。

include ../common/Makefile.workspace
include $(ETROBO_SCRIPTS)/Makefile.fakemake

ここのパスに自分で設定するのか自動で設定されるのかのどちらでしょうか？
ーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーー

分かる方回答御願いします

jtFuruhata, 2020-07-04:
ターミナルの実行ログを全ていただけませんか？　よろしくお願いします。

CIT-stu, 2020-07-06:
●●●●:workspace ○$ make app=sample_c4
Makefile:2: BeerHall/etrobo/scripts/Makefile.fakemake: No such file or directory
make: *** No rule to make target `BeerHall/etrobo/scripts/Makefile.fakemake'.  Stop.

●はPCの名前。
◯はユーザー名。

jtFuruhata, 2020-07-06:
`workspace` から実行していませんか？
(fake)makeは、 `etrobo`上からのみ使用可能です。

CIT-stu, 2020-07-07:
返信ありがとうございます。

etrobo上で実行したらまたエラーが出てきました。以下はエラーコードです。

＜エラー＞
「up」無し
●●:etrobo ◯◯$ make app=sample_c4
make: *** No targets specified and no makefile found.  Stop.

「up」有り
●●:etrobo ◯◯$ make app=sample_c4 up
make: *** No rule to make target `up'.  Stop.

●：PCの名前
◯：ユーザー名

jtFuruhata, 2020-07-07:
`ls -la "$ETROBO_ROOT` の結果を頂けますか？　よろしくお願いします。

CIT-stu, 2020-07-08:
●●:~ ◯◯$ ls -la "$ETROBO_ROOT"
total 88
drwxr-xr-x  22 etr  staff    704  6 24 17:25 .
drwxr-xr-x  14 etr  staff    448  6 22 16:44 ..
-rw-r--r--@  1 etr  staff  10244  6 24 17:54 .DS_Store
drwxr-xr-x  12 etr  staff    384  6 16 13:41 .git
-rw-r--r--   1 etr  staff    476  6 16 13:41 .gitignore
drwxr-xr-x   3 etr  staff     96  6 22 17:39 .vscode
-rw-r--r--   1 etr  staff   1086  6 16 13:41 LICENSE
-rw-r--r--   1 etr  staff  15970  6 16 13:41 README.md
drwxr-xr-x  10 etr  staff    320  6 16 14:00 athrill
drwxr-xr-x   3 etr  staff     96  6  3 15:05 athrill-gcc-package
drwxr-xr-x   9 etr  staff    288  6 16 14:00 athrill-target-v850e2m
drwxr-xr-x   6 etr  staff    192  6 16 14:03 dist
drwxr-xr-x   3 etr  staff     96  6 24 16:32 etrobo2020
drwxr-xr-x@  7 etr  staff    224  6 24 17:23 etroboEV3-master
drwxr-xr-x   8 etr  staff    256  4 14 16:19 ev3rt-1.0-release
drwxr-xr-x  22 etr  staff    704  6 16 14:03 ev3rt-athrill-v850e2m
drwxr-xr-x   6 etr  staff    192  6 16 13:58 gcc-arm-none-eabi-6-2017-q1-update
drwxr-xr-x  24 etr  staff    768  6 24 17:53 hrp3
-rwxr-xr-x@  1 etr  staff     52  6 16 13:41 make
drwxr-xr-x  18 etr  staff    576  6 16 13:41 scripts
-rwxr-xr-x   1 etr  staff     61  6 16 13:41 update
lrwxr-xr-x   1 etr  staff     18  6 16 13:58 workspace -> hrp3/sdk/workspace
MacBook-Pro-Penguin:~ etr$

jtFuruhata, 2020-07-08:
`make` を手動ダウンロードして配置したのではないかと思います、 `ls -l@ make` の結果をいただけますか？
もしも com.apple.quarantine が付いているようであれば、
`xattr -d com.apple.quarantine make` すると動きそうな気がします。

CIT-stu, 2020-07-10:
こちらになります。

●●:etrobo ○○$ ls -l@ make
-rwxr-xr-x@ 1 etr  staff  52  6 16 13:41 make
	com.apple.lastuseddate#PS	16

jtFuruhata, 2020-07-10:
makeをmacOS上から起動した形跡ですね、色々試されたんですね。
とりあえず消しておきますか。あと、この間に、ETロボコン参加者のコミュニティSlackでのフィードバックで更新した部分を適用するため、`update dist`を2回実行します。

```
xattr -d com.apple.lastuseddate#PS make
update dist
update dist
```

これで再度 `make sample`のログを全部ください。

あと、確認いただきたいのは、
- Visual Studio Codeのインストール先
    - `ls -la /usr/local/bin/code` の結果をください。
- workspaceフォルダの中身
    - `ls -la workspace/Makefile` の結果をください。
- sample_c4の中身
    - `ls -la workspace/sample_c4/` の結果をください。
    - 何も表示されない、またはエラーが出た場合は、VSCode左側に④がついたアイコンが出ていないかどうか確認してください。

CIT-stu, 2020-07-14:
make sampleのログです。
ーーーーーーーーーーーーーーーーーーーーーーーーーー
●:etrobo ◯$ make app=sample_c4 up
invoker make app=sample_c4 up
rm -rf /Users/etr/BeerHall/etrobo/hrp3/sdk/workspace/.././OBJ/
Generating Makefile from ../common/Makefile.app.
gmake[1]: ディレクトリ '/Users/etr/BeerHall/etrobo/hrp3/sdk/OBJ' に入ります
rm -f cfg1_out cfg1_out.o cfg1_out.c cfg1_out.syms cfg1_out.srec module_cfg.h module_cfg.c \#* *~ *.o
gmake[1]: ディレクトリ '/Users/etr/BeerHall/etrobo/hrp3/sdk/OBJ' から出ます
gmake[1]: ディレクトリ '/Users/etr/BeerHall/etrobo/hrp3/sdk/OBJ' に入ります
  CFG[1]  module_cfg.h
[cfg.rb] Generated cfg1_out.c
  CFG[2]  module_cfg.h
[cfg.rb] Generated module_cfg.c
[cfg.rb] Generated module_cfg.h
gmake[1]: ディレクトリ '/Users/etr/BeerHall/etrobo/hrp3/sdk/OBJ' から出ます
gmake[1]: ディレクトリ '/Users/etr/BeerHall/etrobo/hrp3/sdk/OBJ' に入ります
  CC      module_cfg.c
  CC      ../../library/t_perror.c
  CC      ../../library/strerror.c
  CC      ../../library/vasyslog.c
  CC      ../../target/ev3_gcc/TLSF-2.4.6/src/tlsf.c
  CC      ../workspace/sample_c4/app.c
  CC      ../common/ev3api/src/ev3api.c
  CC      ../common/ev3api/src/ev3api_battery.c
  CC      ../common/ev3api/src/ev3api_brick.c
  CC      ../common/ev3api/src/ev3api_fs.c
../workspace/sample_c4/app.c: In function 'main_task':
../workspace/sample_c4/app.c:137:21: warning: too many arguments for format [-Wformat-extra-args]
         fprintf(bt, "Bluetooth Remote Start: Ready.\n", EV3_SERIAL_BT);
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../workspace/sample_c4/app.c:138:21: warning: too many arguments for format [-Wformat-extra-args]
         fprintf(bt, "send '1' to start\n", EV3_SERIAL_BT);
                     ^~~~~~~~~~~~~~~~~~~~~
At top level:
../workspace/sample_c4/app.c:64:5: warning: 'gyro_sensor' defined but not used [-Wunused-const-variable=]
     gyro_sensor     = EV3_PORT_4;
     ^~~~~~~~~~~
  CC      ../common/ev3api/src/ev3api_speaker.c
  CC      ../common/ev3api/src/ev3api_lcd.c
  CC      ../common/ev3api/src/ev3api_motor.c
  CC      ../common/ev3api/src/ev3api_newlib.c
  CC      ../common/ev3api/src/ev3api_sensor.c
  LD      app
gmake[1]: ディレクトリ '/Users/etr/BeerHall/etrobo/hrp3/sdk/OBJ' から出ます
'sample_c4' is copied into EV3.
fakemake on HRP3: build succseed: sample_c4

CIT-stu, 2020-07-14:
ls -la /usr/locale/bin/code の結果です。
ーーーーーーーーーーーーーーーーーーーーーーーーーーー
lrwxr-xr-x  1 etr  admin  75  6 19 17:17 /usr/local/bin/code -> /Users/etr/Downloads/Visual Studio Code.app/Contents/Resources/app/bin/code

CIT-stu, 2020-07-14:
ls -la workspace/Makefile の結果です。
ーーーーーーーーーーーーーーーーーーーーーーーーーーーー
-rw-r--r--  1 etr  staff  81  6 16 13:58 workspace/Makefile

CIT-stu, 2020-07-14:
ls -la workspace/sample_c4/ の結果です。
ーーーーーーーーーーーーーーーーーーーーーーーーーーーー
total 64
drwxr-xr-x   8 etr  staff   256  7 14 09:42 .
drwxr-xr-x  14 etr  staff   448  7 14 08:56 ..
-rwxr-xr-x   1 etr  staff    74  7 14 09:42 Makefile.inc
-rwxr-xr-x   1 etr  staff    54  7 14 09:42 Makefile.inc.base
-rwxr-xr-x   1 etr  staff    44  7 14 09:42 Makefile.inc.org
-rwxr-xr-x   1 etr  staff  9093  6 14 15:13 app.c
-rwxr-xr-x   1 etr  staff   255  6 12 09:20 app.cfg
-rwxr-xr-x   1 etr  staff  3892  6 12 09:20 app.h

CIT-stu, 2020-07-14:
make appの方は無事成功できました。
ご指導有難うございました。今後も宜しく御願いします。

jtFuruhata, 2020-07-14:
ログを見る限り、 `xattr -d` でうまくいったようですね、よかったです。

closed 2020-07-14 by jtFuruhata