ETrobocon/etrobo #20: シミュレーター用ビルドでM_PI関数が使えない？

haruki-takashina, 2020-07-14:
今回新たに導入したシミュレーション環境でも
「make app=プロジェクト名」でのビルドでは問題なく#include <math.h>のM_PIが利用できましたが
「make app=プロジェクト名 sim up」でシミュレーター起動をすると
 error: 'M_PI' undeclared (first use in this function)
     curcnt_L=left * M_PI / 180.0;  
   　　　　　　　　^
となって起動に失敗してしまいます。
とりあえず手打ちで円周率を用意してビルドをかけるとシミュレーター上でも動くみたいですが精度が落ちてしまう？みたいです。

実行環境はWindows10のbash上でUbuntuを動かしています。
ロボコン初心者でよくわからないことが多いですが
ご確認お願い致します。

jtFuruhata, 2020-07-14:
プロジェクトフォルダ（sample_c4など）の`Makefile.inc`に以下の1行を追加してみてください。

```
APPL_LIBS += -lm
```

haruki-takashina, 2020-07-17:
Makefile.incに追加してコンパイルしましたがシミュレーター上のコンパイル中でやはり止まってしまいました・・・

jtFuruhata, 2020-07-17:
この件ちょっと追ってみたところ、Athrill側の不具合のように見えます。

- $ETROBO_ROOT/athrill-gcc-package/usr/local/athrill-gcc/v850-elf/include/math.hは見に行っている
- ただし、当該#defineは `#ifndef __STRICT_ANSI__` 内にあって、このフラグが立った状態でビルドが行われているようだ

応急処置として、コードの先頭（例えばsample_c4.cであれば10行目あたり）に以下の1行を追加してください。

```
#undef __STRICT_ANSI__
```

Athrill側の対応が済み次第ここでもお知らせします。

haruki-takashina, 2020-07-17:
ありがとうございます。

ytoi, 2020-07-18:
コンパイラに対して`-std=c99`オプション が指定されているため、ビルドインマクロとしてSTRICT_ANSIが定義されていました。

sdk/workspace/Makefile.img内の66行目と70行目にある

`COPTS += -std=c99`

を消すことで、解除はできます。c99オプションはC++の標準テンプレートが使えるように加えたものですが、これが悪さをしていました。
`COPTS += -std=gnu99`

とすることで、STRICT_ANSIも定義されず、標準テンプレートも使えるようになります。
マスター側で削除するか、gnu99とするかは検討しますが、当面は参加者側で対応をお願いします。

ビルトインマクロについては
`echo | /usr/local/bin/v850-elf-gcc -dM -E -`

のようにすることで見ることができます。
これで
`echo | /usr/local/bin/v850-elf-gcc -dM -E - | grep STRICT`
は何も表示されませんが、
`echo | /usr/local/bin/v850-elf-gcc -dM -E -std=c99 - | grep STRICT`
は
`#define __STRICT_ANSI__ 1`
と表示されます。
`echo | /usr/local/bin/v850-elf-gcc -dM -E -std=gnu99 - | grep STRICT`
とすると表示されません。

closed 2020-07-19 by jtFuruhata