ETrobocon/etrobo #21: etroboパッケージのインストールが出来ません

Naoyamagucih, 2020-07-14:
環境はWindows10です。　etroboパッケージインストール前の動作環境構築手順は完了しています。

コマンドプロンプトにコマンドをコピー&ペーストし実行すると、Start ETrobo.cmdは作成されるのですが、
 その後ダブルクリックしてもインストールを開始どころか反応がありません。

何卒お力添えのほど宜しくお願い致します。

jtFuruhata, 2020-07-14:
HTTPプロキシの内側からインストールしていないでしょうか？
もしもそうでしたら、現在この対策アップデートを作成中です。
もちろん、Ubuntuで適切にプロキシの設定が行われていれば問題は発生しません。

Naoyamagucih, 2020-07-28:
仰る通り、プロキシの設定をしていませんでした。

プロキシを設定後再度実行すると、
インストールを開始しましたが
fatal: unable to access 'https://github.com/ETrobocon/etrobo.git/': Failed to connect to github.com port 443: Connection refused
のエラーが出ました。

jtFuruhata, 2020-07-28:
```
git config --global http.proxy HTTPプロキシ設定URL
git config --global https.proxy HTTPプロキシ設定URL
```

を試してみてください。 `HTTPプロキシ設定URL` は `http_proxy` 環境変数と同じものを指定します。

Naoyamagucih, 2020-07-29:
etroboパッケージがインストールできました。
ありがとうございます。

jtFuruhata, 2020-07-29:
よかったです、実装も急ぎます…

closed 2020-07-29 by jtFuruhata