ETrobocon/etrobo #30: プロジェクトがシンボリックリンクの場合、$ETROBO_ATHRILL_WORKSPACEにシンボリックリンクがコピーされる

Amakuchisan, 2020-10-17:
Windows10で開発しています。

プロジェクトをシンボリックリンクで配置すると、fakemake.shの
https://github.com/ETrobocon/etrobo/blob/19e84318d4927c5bd699c46bd1cfa1199d041054/scripts/fakemake.sh#L161
の箇所で、ディレクトリのコピーではなく、シンボリックリンクをコピーします。

シンボリックリンクをコピーの状態で`make sim up`を行うと、`$ETROBO_HRP3_WORKSPACE/$proj`と`$ETROBO_ATHRILL_WORKSPACE/`のファイルが同じものを指すため、
`$ETROBO_ATHRILL_WORKSPACE`での、`mv -f "${incFile}.org" "$incFile"`がうまくいきません。
`mv: cannot stat '{プロジェクト名}/Makefile.inc.org': No such file or directory`
その結果、実行するたびに、Makefile.incのファイルの末尾に`COPTS += -DMAKE_SIM`が追加されていくといった状態になります。

この問題を解決するために、我々のチームは、fakemake.shの該当の行に-Hオプションを付加しました。
cpコマンドの、-Hオプションを使用すると、対象がシンボリックリンクの場合は、リンク先をファイルやフォルダとしてコピーするようになります。
`cp -rH "$ETROBO_HRP3_WORKSPACE/$proj" "$ETROBO_ATHRILL_WORKSPACE/"`

プロジェクトをシンボリックリンクで管理できるようになると、嬉しいため、検討していただけると幸いです。

https://github.com/ETrobocon/etrobo/blob/19e84318d4927c5bd699c46bd1cfa1199d041054/scripts/fakemake.sh#L161-L169

現状の開発イメージ図
[image]

jtFuruhata, 2020-10-18:
なるほどご提案ありがとうございます。ETロボコン参戦に利用する場合でも、etrobo環境は適宜書き換えて構いませんので、この変更のままお使いいただいて結構です。

レポジトリの修正は、実はそのままでは適用できません…　macOS(Mojave)の標準cpコマンドでは-rHオプションを指定することができません。他にも実装中細かな違いがあって悩ましかったのですが、今年度のMacのBeerHallにはGNU coreutilを含んでおらず、今からHotFixするのも重いため、来年度のBeerHallにはGNU coreutilを含むこととし、その時点でこの変更を適用させていただきたいと考えています。

また、よろしければプルリクエストを出して、contributorの一員となってください。

jtFuruhata, 2021-03-07:
MacのGNU coreutilsへの移行を完了し、上記修正を適用しました。ありがとうございました。

commit e5c1da3c (2021-03-07): fix: #30 unable to use with symbolick projects
closed 2021-03-07 by jtFuruhata