ETrobocon/etrobo #31: プログラムが正常に動作しない

inazuma04157, 2020-11-12:
"要素技術とモデルを開発に使おう: コードとモデル図を対応づけてみよう"のsample00を作成し
シミュレーター上で実行したのですがすぐに止まってしまいます。
＜作業環境＞
windows10

止まった時にはこの文が同時に出ます
[ l_app runner: exit: Athrill2 was shutted down ]
[ l_app runner: duration of running: 8 secs in real-world time ]
[ athrill_runner #19712: terminated. ]
[ asp launcher #19671: terminated. ]

jtFuruhata, 2020-11-13:
上記メッセージは正常終了の場合でも出ます。その他の部分のログもいただけますか？
（ユーザ名など機微情報は黒塗りで結構です）

inazuma04157, 2020-11-13:
こちらで問題ないでしょうか

invoker make app=sample00 sim up
rm -rf /home/ユーザー名/etrobo/hrp3/sdk/workspace/.././OBJ/
/usr/lib/ruby/2.5.0/shell.rb:147: warning: Insecure world writable dir /mnt/c/app/taten/product/11.1.0 in PATH, mode 040777
Generating Makefile from ../common/Makefile.app.
make[1]: Entering directory '/home/ユーザー名/etrobo/hrp3/sdk/OBJ'
rm -f cfg1_out cfg1_out.o cfg1_out.c cfg1_out.syms cfg1_out.srec module_cfg.h module_cfg.c \#* *~ *.o
make[1]: Leaving directory '/home/ユーザー名/etrobo/hrp3/sdk/OBJ'
make[1]: Entering directory '/home/ユーザー名/etrobo/hrp3/sdk/OBJ'
  CFG[1]  module_cfg.h
[cfg.rb] Generated cfg1_out.c
  CFG[2]  module_cfg.h
[cfg.rb] Generated module_cfg.c
[cfg.rb] Generated module_cfg.h
make[1]: Leaving directory '/home/ユーザー名/etrobo/hrp3/sdk/OBJ'
make[1]: Entering directory '/home/ユーザー名/etrobo/hrp3/sdk/OBJ'
  CC      module_cfg.c
  CC      ../../library/t_perror.c
  CC      ../../library/strerror.c
  CC      ../../library/vasyslog.c
  CC      ../../target/ev3_gcc/TLSF-2.4.6/src/tlsf.c
  CC      ../common/ev3api/src/ev3api.c
  CC      ../common/ev3api/src/ev3api_battery.c
  CC      ../common/ev3api/src/ev3api_brick.c
  CC      ../common/ev3api/src/ev3api_fs.c
  CC      ../common/ev3api/src/ev3api_speaker.c
  CC      ../common/ev3api/src/ev3api_lcd.c
  CC      ../common/ev3api/src/ev3api_motor.c
  CC      ../common/ev3api/src/ev3api_newlib.c
  CC      ../common/ev3api/src/ev3api_sensor.c
  CXX     ../workspace/sample00/util.cpp
  CXX     ../workspace/sample00/app.cpp
  LD      app
make[1]: Leaving directory '/home/ユーザー名/etrobo/hrp3/sdk/OBJ'
fakemake on HRP3: build succeed: sample00
rm -rf /home/ユーザー名/etrobo/ev3rt-athrill-v850e2m/sdk/workspace/.././OBJ/
cd /home/ユーザー名/etrobo/ev3rt-athrill-v850e2m/sdk/workspace/.././OBJ/ && \
ruby /home/ユーザー名/etrobo/ev3rt-athrill-v850e2m/sdk/workspace/../../configure.rb  -T v850_gcc -A app \
        -a ../workspace/sample00 \
        -D ../.. \
        -l c++ \
        -m ../common/Makefile.img \
        -o "" \
        -t \
        -d ./dep \
        -C ../common/athrill_common.cdl \
        -U "" && \
make clean
/usr/lib/ruby/2.5.0/shell.rb:147: warning: Insecure world writable dir /mnt/c/app/taten/product/11.1.0 in PATH, mode 040777
Generating Makefile from ../common/Makefile.img.
make[1]: Entering directory '/home/ユーザー名/etrobo/ev3rt-athrill-v850e2m/sdk/OBJ'
rm -f \#* *~ ./*.o ./dep/*.d  check.timestamp
rm -f asp asp.syms asp.srec asp.bin
rm -f kernel_opt.timestamp   cfg3_out.db
rm -f cfg3_out.syms cfg3_out.srec 
rm -f kernel_cfg.timestamp kernel_cfg.h kernel_cfg.c   cfg2_out.db
rm -f cfg2_out.syms cfg2_out.srec kernel_cfg_asm.S 
rm -f offset.timestamp offset.h
rm -f cfg1_out.syms cfg1_out.srec cfg1_out
rm -f cfg1_out.timestamp cfg1_out.c cfg1_out.db
rm -rf ./gen
rm -f libkernel.a
make[1]: Leaving directory '/home/ユーザー名/etrobo/ev3rt-athrill-v850e2m/sdk/OBJ'
# -S "banner.o syslog.o serial.o logtask.o" 
(cd /home/ユーザー名/etrobo/ev3rt-athrill-v850e2m/sdk/workspace/.././OBJ/ && \
make offset.h kernel_cfg.h && \
make -j8 > /dev/null ) && \
cp /home/ユーザー名/etrobo/ev3rt-athrill-v850e2m/sdk/workspace/.././OBJ//asp /home/ユーザー名/etrobo/ev3rt-athrill-v850e2m/sdk/workspace && \
echo "/home/ユーザー名/etrobo/ev3rt-athrill-v850e2m/sdk/workspace/sample00" > /home/ユーザー名/etrobo/ev3rt-athrill-v850e2m/sdk/workspace/appdir
[... 129 lines cut ...]
ELF STRING TABLE SECTION LOADED:index=94
DEBUG_FUNC_VDEV_TX_IPADDR = 127.0.0.1
VDEV:TX IPADDR=127.0.0.1
VDEV:TX PORTNO=54001
VDEV:RX IPADDR=127.0.0.1
VDEV:RX PORTNO=54002
DEVICE_CONFIG_COMPLEMENTAL_TX_SENDING=1
DEVICE_CONFIG_RESET_AREA_OFFSET=68
DEVICE_CONFIG_RESET_AREA_SIZE=20
DEBUG_FUNC_FT_LOG_SIZE=1024
athrill_device_func_call=0x200524

TOPPERS/ASP3 Kernel Release 3.2.0 for V850-ESFK3 (Nov 13 2020, 11:40:22)
Copyright (C) 2000-2003 by Embedded and Real-Time Systems Laboratory
                            Toyohashi Univ. of Technology, JAPAN
Copyright (C) 2004-2017 by Embedded and Real-Time Systems Laboratory
            Graduate School of Information Science, Nagoya Univ., JAPAN

brick_dri initialized.
   _____   ______ ___  ______
  / __/ | / /_  // _ \/_  __/
 / _/ | |/ //_ </ , _/ / /
/___/ |___/____/_/|_| /_/
 
====================>Beta 7<=

jtFuruhata, 2020-11-13:
惜しい…その続きが…みたい…です…
EV3RTの起動を確認したメッセージが出ていると思うのですが。

inazuma04157, 2020-11-15:
返信遅れてしまってすいません、こちらで問題ないでしょうか

Powered by TOPPERS/ASP3 RTOS of Hakoniwa
Initialization is completed..
System logging task is started.
[ l_app runner: V850E2M VCPU is powered on ]
[ l_app runner: Athrill2 is booted up ]
[ launcher: CONTROLLABLE detected ]

*** WARNING ***

You are *NOT* using the 'etrobosim2020_4.0' which is a simulator for competition.
If you are a participant, run 'update sim' to download it.

[ l_app runner: EV3RT is booted up ]
[ l_app runner: EV3RT boot time: 7 secs. ]

[ l_app runner: exit: Athrill2 was shutted down ]
[ l_app runner: duration of running: 10 secs in real-world time ]
[ athrill_runner #855: terminated. ]
[ asp launcher #811: terminated. ]

jtFuruhata, 2020-11-15:
なるほど、これはWARNINGメッセージの通りです、文面を拝見したところETロボコンに参加されている方の様ですので、指示に従ってシミュレータをダウンロードしてインストールしてください。

inazuma04157, 2020-11-15:
update simを実行したところユーザーネームとパスワードを要求されましたが何を入力すればいいのでしょうか？

jtFuruhata, 2020-11-15:
ETロボコン参加者に割り当てられているサイボウズのアカウントです。

inazuma04157, 2020-11-15:
無事動きました、お手数おかけ致しました

closed 2020-11-15 by jtFuruhata