ETrobocon/etrobo #41: Macの開発環境構築が失敗する

grace2riku, 2023-03-06:
# 問い合わせ内容
Macの開発環境構築が失敗してしまいます。

なにかお気づきの点があればご教示ください。よろしくお願いいたします。

 * 【環境構築作業手順】の手順で環境構築を試しました。
 * 【環境構築作業手順】の手順で環境構築を試しましたが、「環境構築の前提条件を満たしていないかもしれない」、
という不安があるので【事前質問】に記載しました。こちらもご確認いただけますとありがたいです。

# PC環境
## MacBook
* 1.2 GHz デュアルコアIntel Core m3
* メモリ: 8GB
* macOS: Ventura 13.2.1

## Xcodeインストール状況
brew --configコマンドでCLT(Command Line Tools)のバージョンを確認しました。
実行結果を引用・抽出しました。

```
brew --config
CLT: 11.0.0.33.8
Xcode: N/A
```

※[こちらの参考ページ](https://techracho.bpsinc.jp/wingdoor/2021_04_09/104821)でいうところのシステムのCLT(Command Line Tools)のみ存在している状態です。

# 事前質問
こちらの[etroboパッケージインストール前の動作環境構築手順（macOS）](https://github.com/ETrobocon/etrobo/wiki/sim_mac)に

 **HomeBrewを用いたツールチェーンサンドボックス「BeerHall」をホームディレクトリに構築し・・・** とありますが、具体的に何をすれば良いでしょうか?

ホームディレクトリにBeerHallディレクトリをつくれば良いのでしょうか?

とくに環境構築の手順が記載されていないので[etroboパッケージインストール前の動作環境構築手順（macOS）](https://github.com/ETrobocon/etrobo/wiki/sim_mac)のページに書いてある

 * Visual Studio Codeのインストール
 * Visual Studio Codeインストール後の設定（インストール済みの人も）

を実施し終了しています。

# 環境構築作業手順
1. [こちらのページ](https://github.com/ETrobocon/etrobo/wiki/etrobo_pkg_install_mac)のコマンドをターミナルを開き実行した。

コマンド：

```
cd Desktop; echo 'name=startetrobo_mac.command; if [ ! -f $name ]; then curl -O https://raw.githubusercontent.com/ETrobocon/etrobo/master/scripts/$name; chmod +x ~/$name; fi; ~/$name' > "Start ETrobo.command"; chmod +x "Start ETrobo.command"
```

2. デスクトップに【Start ETrobo.command】が作成される。

Start ETrobo.commandの中身はつぎのとおり：

```
name=startetrobo_mac.command; if [ ! -f $name ]; then curl -O https://raw.githubusercontent.com/ETrobocon/etrobo/master/scripts/$name; chmod +x ~/$name; fi; ~/$name
```

3. デスクトップに作成されたStart ETrobo.commandをダブルクリックする

4. ターミナルにつぎが表示される。

ログ内容を見る限り必要なディレクトリ、ファイルが存在していないようです。

ターミナルのログ表示：

```
Last login: Mon Mar  6 22:17:17 on ttys003

The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
/Users/k-abe/Desktop/Start\ ETrobo.command ; exit;
(base) KojinoMacBook-2:~ k-abe$ /Users/k-abe/Desktop/Start\ ETrobo.command ; exit;

------------
 jtBeerHall - an implementation of Homebrew sandbox
------------
 as 'startetrobo_mac.command' Ver 5.25a.220617
make aliase to code
/Users/k-abe/startetrobo_mac.command: line 371: /Users/k-abe/BeerHall/usr/local/bin/code: No such file or directory
chmod: /Users/k-abe/BeerHall/usr/local/bin/code: No such file or directory
make symbolic link from $BEERHALL/etc to $HOMEBREW_PREFIX/etc
ln: /Users/k-abe/BeerHall/etc: No such file or directory
ls: /usr/local/lib/libfl*: No such file or directory
make symbolic link from /usr/local/lib to flex/lib
ls: /Users/k-abe/BeerHall/usr/local/opt/flex/lib: No such file or directory
/Users/k-abe/startetrobo_mac.command: line 466: cd: /Users/k-abe/BeerHall: No such file or directory
bash on BeerHall is invoked on /Users/k-abe
invoker params: 
/Users/k-abe/startetrobo_mac.command: line 475: /Users/k-abe/BeerHall/BeerHall: No such file or directory
logout

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[プロセスが完了しました]
```

khachiman, 2023-03-07:
>ホームディレクトリにBeerHallディレクトリをつくれば良いのでしょうか?

BeerHall ディレクトリは、Start ETrobo.command 実行時に自動で作成されます。

>/Users/k-abe/startetrobo_mac.command: line 466: cd: /Users/k-abe/BeerHall: No such file or directory

ということなので、BeerHall ディレクトリ が作成されていないようです。

この場合は、インストール作業を最初から行った方がよいです

[etroboパッケージのインストールが継続出来なくなってしまった。](https://github.com/ETrobocon/etrobo/wiki/install_error#etrobo%E3%83%91%E3%83%83%E3%82%B1%E3%83%BC%E3%82%B8%E3%81%AE%E3%82%A4%E3%83%B3%E3%82%B9%E3%83%88%E3%83%BC%E3%83%AB%E3%81%8C%E7%B6%99%E7%B6%9A%E5%87%BA%E6%9D%A5%E3%81%AA%E3%81%8F%E3%81%AA%E3%81%A3%E3%81%A6%E3%81%97%E3%81%BE%E3%81%A3%E3%81%9F%E3%80%82) の対応か、[BeerHallも含めた全体のアンインストール/再インストール](https://github.com/ETrobocon/etrobo/wiki/etrobo_pkg_update_etc#beerhall%E3%82%82%E5%90%AB%E3%82%81%E3%81%9F%E5%85%A8%E4%BD%93%E3%81%AE%E3%82%A2%E3%83%B3%E3%82%A4%E3%83%B3%E3%82%B9%E3%83%88%E3%83%BC%E3%83%AB%E5%86%8D%E3%82%A4%E3%83%B3%E3%82%B9%E3%83%88%E3%83%BC%E3%83%AB)を方がよいです。個人的には後者をお勧めします。

grace2riku, 2023-03-07:
@khachiman 
ありがとうございます。どうしてよいか全くわからなかったのでとても助かりました。

[BeerHallも含めた全体のアンインストール/再インストール](https://github.com/ETrobocon/etrobo/wiki/etrobo_pkg_update_etc#beerhall%E3%82%82%E5%90%AB%E3%82%81%E3%81%9F%E5%85%A8%E4%BD%93%E3%81%AE%E3%82%A2%E3%83%B3%E3%82%A4%E3%83%B3%E3%82%B9%E3%83%88%E3%83%BC%E3%83%AB%E5%86%8D%E3%82%A4%E3%83%B3%E3%82%B9%E3%83%88%E3%83%BC%E3%83%AB)に記載の手順を実施し少し開発環境構築が進みました。

しかし構築後にデスクトップのStart ETrobo.commandを実行してもつぎのログが表示され、etrobo環境のvscodeが起動しませんでした。

```
Last login: Tue Mar  7 12:35:48 on ttys003

The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
/Users/k-abe/Desktop/Start\ ETrobo.command ; exit;
(base) KojinoMacBook-2:~ k-abe$ /Users/k-abe/Desktop/Start\ ETrobo.command ; exit;

------------
 jtBeerHall - an implementation of Homebrew sandbox
------------
 as 'startetrobo_mac.command' Ver 5.25a.220617
ls: /usr/local/lib/libfl*: No such file or directory
make symbolic link from /usr/local/lib to flex/lib
ls: /Users/k-abe/BeerHall/usr/local/opt/flex/lib: No such file or directory
bash on BeerHall is invoked on /Users/k-abe/BeerHall
invoker params: 
Welcome, you are in jtBeerHall - an implementation of Homebrew sandbox
bash on BeerHall is invoked by params: ./startetrobo  
bash: ./startetrobo: No such file or directory
logout

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[プロセスが完了しました]
```

開発環境構築中はつぎログが多数表示されていました。

```
Error: Your Command Line Tools are too outdated.
Update them from Software Update in System Preferences.

If that doesn't show you any updates, run:
  sudo rm -rf /Library/Developer/CommandLineTools
  sudo xcode-select --install

Alternatively, manually download them from:
  https://developer.apple.com/download/all/.
You should download the Command Line Tools for Xcode 14.1.
```

Command Line Toolsをアップデートしてもう一度、環境構築をやり直してみようと思います。

# 開発環境構築中のターミナルのログ
```
Last login: Tue Mar  7 11:47:25 on ttys002
/Users/k-abe/Desktop/Start\ ETrobo.command ; exit;

The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
(base) KojinoMacBook-2:~ k-abe$ /Users/k-abe/Desktop/Start\ ETrobo.command ; exit;
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 15741  100 15741    0     0  35459      0 --:--:-- --:--:-- --:--:-- 35452

------------
 jtBeerHall - an implementation of Homebrew sandbox
[... 1811 lines cut ...]

make aliase to code
make symbolic link from $BEERHALL/etc to $HOMEBREW_PREFIX/etc
ls: /usr/local/lib/libfl*: No such file or directory
make symbolic link from /usr/local/lib to flex/lib
ls: /Users/k-abe/BeerHall/usr/local/opt/flex/lib: No such file or directory
make BeerHall
download startetrobo
/Users/k-abe/startetrobo_mac.command: line 459: /Users/k-abe/BeerHall/usr/local/bin/wget: No such file or directory
chmod: startetrobo: No such file or directory
bash on BeerHall is invoked on /Users/k-abe/BeerHall
invoker params: 
Welcome, you are in jtBeerHall - an implementation of Homebrew sandbox
bash on BeerHall is invoked by params: ./startetrobo  
bash: ./startetrobo: No such file or directory
logout

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
Deleting expired sessions...none found.

[プロセスが完了しました]
```

khachiman, 2023-03-07:
> Error: Your Command Line Tools are too outdated.
> Update them from Software Update in System Preferences.
>
> If that doesn't show you any updates, run:
>  sudo rm -rf /Library/Developer/CommandLineTools [...]

Command Line Tools for Xcode のバージョンが14.1が必要なようです。(14.1以上であれば大丈夫だと思います)
すでにアップデートは試みたようですが、いくつになっていますか?

grace2riku, 2023-03-07:
@khachiman 
ありがとうございます。
Command Line Tools for Xcode のバージョンは14.2.0.0.1.1668646533にアップデートしました。

その後、[BeerHallも含めた全体のアンインストール/再インストール](https://github.com/ETrobocon/etrobo/wiki/etrobo_pkg_update_etc#beerhall%E3%82%82%E5%90%AB%E3%82%81%E3%81%9F%E5%85%A8%E4%BD%93%E3%81%AE%E3%82%A2%E3%83%B3%E3%82%A4%E3%83%B3%E3%82%B9%E3%83%88%E3%83%BC%E3%83%AB%E5%86%8D%E3%82%A4%E3%83%B3%E3%82%B9%E3%83%88%E3%83%BC%E3%83%AB)を実行しアンインストールしました。

再度、開発環境構築をはじめから行い2時間経過・・・という状況です。
いまのところ、前回の環境構築で発生していたつぎのログは発生していないです。

```
Error: Your Command Line Tools are too outdated.
Update them from Software Update in System Preferences.

If that doesn't show you any updates, run:
  sudo rm -rf /Library/Developer/CommandLineTools
  sudo xcode-select --install

Alternatively, manually download them from:
  https://developer.apple.com/download/all/.
You should download the Command Line Tools for Xcode 14.1.
```

なにか進展ありましたら更新させていただきます。

jtFuruhata, 2023-03-07:
BeerHallの段階、HomeBrewでのパッケージのインストールに失敗しているような感じです。anaconda系のものがインストールされているなどあまりクリーンな.bashrcではないようで、HomeBrewのインストール先が他に奪われてしまっているのかもしれません。
（ `/Users/k-abe/startetrobo_mac.command: line 459: /Users/k-abe/BeerHall/usr/local/bin/wget: No such file or directory
chmod: startetrobo: No such file or directory
bash on BeerHall is invoked on /Users/k-abe/BeerHall` あたりが致命的ですね）
こちらでもちょっと詳しく追ってみます。

jtFuruhata, 2023-03-07:
なるほど、ログを見る限りHomeBrewパッケージの依存関係がものすごく複雑になっていますね、循環依存しているように見えるものもあって本当に大丈夫かなって気もしているのですが... ちょっと今Macで作業中でBeerHallを全消しできないので少々お時間をいただければと思います。

grace2riku, 2023-03-07:
@jtFuruhata @khachiman 
メッセージいただきありがとうございます。

Command Line Tools for Xcode のバージョンを14.2.0.0.1.1668646533にアップデートした後に、はじめからインストールし直したところ、サンプル(sample_c4)のmake, シミュレータの動作確認ができました。

丁寧にサポートいただきありがとうございます🙇

Macで環境構築する場合は、Command Line Tools for Xcode のバージョンは14.1以上にした方が良さそうですね。

grace2riku, 2023-03-07:
つぎの手順で環境構築が成功したのでまとめます。

 * [BeerHallも含めた全体のアンインストール/再インストール](https://github.com/ETrobocon/etrobo/wiki/etrobo_pkg_update_etc#beerhall%E3%82%82%E5%90%AB%E3%82%81%E3%81%9F%E5%85%A8%E4%BD%93%E3%81%AE%E3%82%A2%E3%83%B3%E3%82%A4%E3%83%B3%E3%82%B9%E3%83%88%E3%83%BC%E3%83%AB%E5%86%8D%E3%82%A4%E3%83%B3%E3%82%B9%E3%83%88%E3%83%BC%E3%83%AB)でアンインストールを実施した。

 * Command Line Tools for Xcode のバージョンを11.0.0.33.8から14.2.0.0.1.1668646533にアップデートした。つぎのコマンド実行した。
```
sudo rm -rf /Library/Developer/CommandLineTools
sudo xcode-select --install
```

* [etroboパッケージ macOSのインストール手順](https://github.com/ETrobocon/etrobo/wiki/etrobo_pkg_install_mac)を実施した。

サポート、ありがとうございました🙇

closed 2023-03-07 by grace2riku