ETrobocon/etrobo #44: ソースファイルを複数のディレクトリに分けて開発したい

grace2riku, 2023-10-05:
こんにちは。お世話になります。
ソースファイルを複数のディレクトリに分けて開発したいと考えています。

https://afrel.co.jp/technology-info/information-ev3/ev3-software/ev3-c-language/44097/

上のページを参考にして、sample_c4をベースにやってみましたがapp.cをmakeするルールがないとなりmakeが失敗しました。

<sample_c4のデフォルトのディレクトリ構成>
```
sample_c4
	Makefile.inc
	app.c
	app.cfg
	app.h
```

<今回試したディレクトリ構成>
```
sample_c4
	Makefile.inc
	test
	    └ app.c
	app.cfg
	app.h
```

testディレクトリの中にapp.cを格納しました。
参考ページのつぎの1, 2の対応を実施しましたが結果はかわらずapp.cをmakeするルールがないとなりmakeが失敗しました。

/hrp3/sdk/common/Makefile.prj.commonのつぎの追加を行う。
1. 
```
APPL_DIR += $(foreach dir,$(shell find $(APPLDIR) -type d),$(dir))
```

2. 
```
APPL_DIR += $(foreach dir,$(shell find $(APPLDIR) -type d -name test),$(dir))
```

※testは今回作成したディレクトリ名です。
※試しているEV3RTの環境は最新のリリース1.1です(ev3rt-1.1-release.zipがダウンロードされていたのでそう判断しました)。

ソースファイルを複数のディレクトリに分けて開発したご経験ある方、なにかご存じの方、対策についてご教示いただけるとありがたいです。
よろしくお願いいたします。

AlfortVanilla, 2023-10-05:
@grace2riku さん、こんにちは。

自分も詳しくはありませんが、、、
今回お試しのtestフォルダを追加した場合ですが、
sample_c4ディレクトリ内のMakefile.incに
以下の追加をしてからmake コマンドを実行したら、いかがでしょうか？
（今回はインクルードパス（INCLUDES+=\～）以降は追加しなくてもいいかも）

```
mkfile_path := $(dir $(lastword $(MAKEFILE_LIST)))
APPL_DIRS += $(mkfile_path)test

INCLUDES += \
	-I$(mkfile_path)test 
```

grace2riku, 2023-10-05:
@AlfortVanilla 
メッセージありがとうございます。教えていただいた修正を試したみたところmakeできました。

参考ページより教えていただいた修正の方がアプリケーションのディレクトリの外のMakefileに変更が発生しないので、
個人的に嬉しいです。

助かりました。また、大変学びになりました。
ありがとうございました。

AlfortVanilla, 2023-10-05:
上手く動いたようで良かったです。
開発がんばってください。（Closeします）

closed 2023-10-05 by AlfortVanilla