ETrobocon/etrobo #45: make sampleが立ち上がりません。

nino-shotaro, 2024-05-26:
初めまして。
ETロボコンの環境構築を行っているのですが、うまくいかないので質問させてください。

当方の環境は下記のとおりです。
12th Gen Intel(R) Core(TM) i9-12900H   2.50 GHz
32.0 GB (31.7 GB 使用可能)
64 ビット オペレーティング システム、x64 ベース プロセッサ
Windows 11 Home

WindowsUpdateは最新にしてあります。

下記の内容に従って環境構築を行いました。
https://github.com/ETrobocon/etrobo/wiki
インストール前の動作環境構築を完了し、インストール手順を完了し、
最後のVscodeが立ち上がった後に、make sampleと入力するとエラーになります。

その際のエラーメッセージは下記のとおりです。

nino@RIO:~/etrobo$ make sample
/home/nino/etrobo/scripts/fakemake.sh: line 10: cd: /home/nino/etrobo/hrp3/sdk/workspace: No such file or directory
/home/nino/etrobo/scripts/fakemake.sh: line 10: cd: /home/nino/etrobo/hrp3/sdk/workspace: No such file or directory
cp: cannot stat 'sample_c4/Makefile.inc': No such file or directory
cp: cannot stat 'sample_c4/Makefile.inc': No such file or directory
cat: sample_c4/Makefile.inc: No such file or directory
invoke: make app=sample_c4 sim up（以降同じ内容が継続）

エラー時の画像
[image]

この問題について、過去の投稿を探してみました。
 ./startetrobo cleanを行って、再度インストールするという投稿があり、数回試してみましたが状況は改善しませんでした。

ここで手詰まりとなっており、困っております。
どのようにすれば解決するのがご教授いただけると助かります。

よろしくお願いいたします。

AlfortVanilla, 2024-07-03:
貼り付けていただいた画像を見ると、ファイルがいろいろ無いようなので
インストールに失敗していると思われます。

参考に、`make sample`が動く場合の画像を・・・。
[image]

記載していただいた内容からは上手くいかない原因は解りませんが
インストールに失敗する要因はいくつかあります。

・インストール開始時、Ubuntuのパスワードを間違えて入力した
・インストール用のパッケージをダウンロードする際にウィルス対策ソフトがブロックしている
・インターネット接続が不安定な場合
・Windowsのユーザー名が日本語

などが考えられますが、いかがでしょうか。

jtFuruhata, 2025-05-20:
質問者の過失であると判断しました。

closed 2025-05-20 by jtFuruhata