ETrobocon/etrobo #52: [Mac]環境構築に失敗する

Reiki1040, 2025-06-26:
## Macでの環境構築について

Mac上でETロボコン開発環境の構築を行っております。
インストール後、VSCodeが起動されません。  
[こちらの記事](https://github.com/ETrobocon/etrobo/issues/41)と症状が酷似しており、参考にして以下の内容を実施しましたが、改善されません。

### 実施した内容

- BeerHall を含めた一式をアンインストール／再インストール :
  ```bash
  ./startetrobo_mac.command clean
  rm ./startetrobo_mac.command
  exit
  ```
---

- Command Line Tools for Xcode を以下のコマンドでアップデート・バージョン確認：
  ```bash
  sudo rm -rf /Library/Developer/CommandLineTools
  sudo xcode-select --install
  brew --config
  ```

   brew --configコマンド出力(抜粋) :
  ```
  CLT: 16.4.0.0.1.1747106510
  Xcode: N/A
  ```

## インストール後、StartETrobo.commandをダブルクリックした後の出力

```
Last login: Thu Jun 26 19:20:49 on ttys003
You have new mail.
/Users/reiki/Desktop/Start\ ETrobo.command ; exit;
(base) reiki@echigos-MacBook-Air ~ % /Users/reiki/Desktop/Start\ ETrobo.command ; exit;

------------
 jtBeerHall - an implementation of Homebrew sandbox
------------
 as 'startetrobo_mac.command' Ver 5.25a.220617
ls: /usr/local/lib/libfl*: No such file or directory
make symbolic link from /usr/local/lib to flex/lib
ls: /Users/reiki/BeerHall/usr/local/opt/flex/lib: No such file or directory
bash on BeerHall is invoked on /Users/reiki/BeerHall
invoker params: 
Welcome, you are in jtBeerHall - an implementation of Homebrew sandbox
ls: /Users/reiki/BeerHall/etc/profile.d: No such file or directory
bash on BeerHall is invoked by params: ./startetrobo  
bash: 行 1: ./startetrobo: No such file or directory

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[プロセスが完了しました]

```

ytoi, 2025-06-26:
MacOSのバージョンはいくつでしょうか。

Reiki1040, 2025-06-27:
rubyを3.2で指定してインストールをしたところ環境構築に成功しました。
しかしながら、`make app=sample_c4 sim up`を実行しても　 
 `make: *** No rule to make target `sim',  Stop.` と表示され、シミュレータが動きません。

Reiki1040, 2025-06-27:
[image]

jtFuruhata, 2025-06-27:
khachimanさん検証ありがとうございます、大分Homebrew各フォーミュラの依存関係が変化したようで、整理して作り直したいと思います。

Reiki1040, 2025-06-27:
ターミナル、VSCodeの再起動、mac本体の再起動を実行し、再度etrobo環境を起動しました。
しかしながら、先ほどの画像と同じ症状が出ます。。。。

jtFuruhata, 2025-06-27:
Reiki1040さん、それは起動ログではありません、起動後の画面です。

Reiki1040, 2025-06-27:
失礼しました。起動ログは以下のようになります。
```
Last login: Fri Jun 27 12:56:08 on ttys000
You have new mail.
/Users/reiki/Desktop/StartETrobo.command ; exit;
(base) reiki@echigos-MacBook-Air ~ % /Users/reiki/Desktop/StartETrobo.command ; exit;

------------
 jtBeerHall - an implementation of Homebrew sandbox
------------
 as 'startetrobo_mac.command' Ver 5.25a.220617
bash on BeerHall is invoked on /Users/reiki/Desktop/BeerHall
invoker params: 
Welcome, you are in jtBeerHall - an implementation of Homebrew sandbox
bash on BeerHall is invoked by params: ./startetrobo  
-----------------
 startetrobo 3.0 - etrobo all-in-one package installer/invoker
-----------------
detect platform: mac.arm64(Homebrew 4.5.8-89-g415c0de on jtBeerHall)
entering SPIKE mode
-----------
 etrobopkg - etrobo package manager
-----------
  manifest version 2025.05.08a

cached athrill-gcc-package-mac_arm64.tar.gz ... checking with manifest
athrill-gcc-package-mac_arm64.tar.gz is already up to date: size=35857495

cached cli-kintone_v1.10.13_macos.zip ... checking with manifest
cli-kintone_v1.10.13_macos.zip is already up to date: size=18762173

cached etrobosim2025_7.0.1-public_mac.tar.gz ... checking with manifest
etrobosim2025_7.0.1-public_mac.tar.gz is already up to date: size=43089544

cached etrobosim2025_7.0.2_mac.tar.gz ... checking with manifest
etrobosim2025_7.0.2_mac.tar.gz is already up to date: size=0
Current status of this Athrill-related repositories:
       athrill: toppers/8bf811e7/master
athrill-target: toppers/dff01d7/master
 raspike-athrill: ETrobocon/95c2071/master
the ETrobo official certified commit
Bundled Simulator: 2025_7.0.1-public

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[プロセスが完了しました]

```

jtFuruhata, 2025-06-27:
Start ETrobo.commandの末尾に「 shell」を追加して「~/$name shell」にして実行して、そのターミナルのログの全部ください（見たいのは最後のプロンプト）

jtFuruhata, 2025-06-27:
プロンプトの出方の違いが分かりますかね？　これが正しく起動している状態です。
それ以外の時はおかしな起動をしているので、どんなにmakeを叩いても無駄です。
ここから `code .` を２回ぐらい繰り返すとVSCodeを直接起動できるようになるんじゃないかと思うんですが、私は面倒なので「設定」から「terminal session」で検索して出てくる「Terminal > Integrated: Enable Persistent Sessions」をオフにしています。

jtFuruhata, 2025-06-27:
もうログは貼っていただかなくて結構です、私は「環境起動後の、あなたの操作に問題がある」、と言っているのです。もしもETロボコンの参加者であれば、ここから先は正規のルートでサポートを受けてください。参加者ではない場合、使い方についてのサポートをする責務を私は負いません。

jtFuruhata, 2025-06-27:
Homebrewフォーミュラの依存関係を最新のものに更新しました。問題は解決したのでクローズします。

closed 2025-06-27 by jtFuruhata